MosaicManufacturing/canvas-issues #103: Unable to create printer profile filamentDiameter required error

ardenpm, 2019-06-28:
**Describe the bug**
When adding a new printer profile, even if filling out all available fields on submitting the error `Parameter 'filamentDiameter' is required but was not provided` is given in an alert box.

**To Reproduce**
Steps to reproduce the behavior:
1. Add a custom printer profile (in my case I had no existing printers)
2. Fill out the information with valid values
3. Submit
4. Observe the above mentioned error.

**Expected behavior**
The profile should be added.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Operating System (please complete the following information):**
 - OS: Windows 10
 - Browser Chrome
 - Version 75.0.3770.100

**Additional context**
I assume this is somehow related to work being done to support 3mm filament.

brandonbloch, 2019-06-28:
Hi @ardenpm,

I'm unable to reproduce this issue on my end. Just want to confirm that you're seeing this on the updated CANVAS UI that was released earlier this week, correct?

[image]

If you're still seeing the old black-and-white UI instead, your browser cache needs to be cleared (this should have already happened by now, but it's possible it hasn't). If the error is in fact happening with the updated version, then it's a different issue entirely and we can go from there.

Cheers,
Brandon

ardenpm, 2019-06-29:
That seems to have been it, it was the older user interface. Working now.

closed 2019-07-04 by brandonbloch