MosaicManufacturing/canvas-issues #104: fan not turning on after first layer Creality CR10S

NZSPY, 2019-06-28:
**Describe the bug**
part cooling Fan doesn't turn on after first layer is printed

**To Reproduce**
slice a file in Canvas
send to printer
observe the fan not coming on after layer 1

**Expected behavior**
The part cooling fan to turn on after layer 1

**Screenshots**
[image]

**Operating System (please complete the following information):**
 - OS: MacOS 10.14.5
 - Browser safari
 - Version 12.1.1

**Additional context**
have attached gcode file, searching file can't find an M106 255 at start of layer 2
[201901MosaicKeychain1.mcf.gcode.txt](https://github.com/MosaicManufacturing/canvas-issues/files/3339162/201901MosaicKeychain1.mcf.gcode.txt)

brandonbloch, 2019-06-28:
Hi NZSPY,

Looking through the GCode there's definitely a bug, but I'm unable to reproduce this with my own printer profiles/projects. Are you using any custom materials, and if so, do they contain any setting overrides (related to the cooling fan or otherwise)? If so, that may be a factor.

Would you mind posting a sharing link to this project, that includes the models as well as the printer/style profile (and custom materials, if present), which we can use to reproduce and fix the bug?

NZSPY, 2019-06-28:
Hi Brandon, 

Here is the ink to my project. Not using any custom materials yet (was playing with adding PVA for supports but haven’t used it yet)

https://share.canvas3d.io/projects/f40e7dc21e6f87a7139da007025ed165 <https://share.canvas3d.io/projects/f40e7dc21e6f87a7139da007025ed165> 

The option to share the materials was greyed out.

Many thanks
Simon

brandonbloch, 2019-07-05:
Hi again,

I believe we've fixed the problem. Of all things it was related to the % sign in some of the fan-related comments! That issue has been resolved and the update will be pushed out within an hour. As I was able to reproduce the problem with my copy of your project, and can no longer reproduce it, I'm marking this issue as resolved. Please re-open it if there are still any problems.

Cheers,
Brandon

closed 2019-07-05 by brandonbloch

NZSPY, 2019-07-05:
Thanks Brandon, 

Great work, yes my fan is now turning on as expected 

Regards
Simon Young 
021 912433