MosaicManufacturing/canvas-issues #106: Cycling print cooling fan

uilleann, 2019-06-29:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Operating System (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

brandonbloch, 2019-07-05:
Hi @uilleann,

This definitely sounds like a bug but I just want to confirm whether you have "Perimeter fan speed" set to "Auto" or a manual value. Would you mind sharing the project (and checking all boxes) so we can try to reproduce the issue?

brandonbloch, 2019-07-05:
After investigating #104, this may be a potential duplicate of that issue. It should be resolved and the fix is being pushed out at the moment, so I'd actually suggest to simply try re-slicing the project in about an hour and see if the correct fan commands are being inserted.

uilleann, 2019-07-05:
It is working now.... Thanks!

brandonbloch, 2019-07-05:
Fantastic, glad to hear that!

closed 2019-07-05 by brandonbloch