MosaicManufacturing/canvas-issues #107: Gap fill is filling gaps outside of print

nallath, 2019-07-04:
**Describe the bug**
As a test, I tried to slice the attached bridge test object, but it seems to be creating gap fill lines outside of the print. 

**To Reproduce**
1. Load the bridge test model.
2. Select the top areas to be printed in a different color
3. Slice and check layerview

**Expected behavior**
No gap filling on the outside of the print and smooth lines. 

**Screenshots**
Preview:
[image]
Closeup:
[image]
Closeup2:
[image]
Linetype closeup:
[image]

**Additional context**
It also seems to be adding quite a "messy" set of line, that could probably be a lot smoother (and thus decrease the jitter of the machine for not having to change direction as often)

[Bridge_Test.zip](https://github.com/MosaicManufacturing/canvas-issues/files/3359512/Bridge_Test.zip)

brandonbloch, 2019-07-05:
Hi @nallath,

This is strange indeed! I'm wondering if this bug is specific to toolpaths created for painted prints—can you try duplicating the project or model and clearing the paint data (drag-and-drop a color onto it outside of the painting UI) and see if the paths are any different?

nallath, 2019-07-15:
I've only got this to behave like this on painted prints, so I'm guessing it's something in that part that's causing it.