MosaicManufacturing/canvas-issues #109: Artifacts when slicing

maximuss, 2019-07-10:
I have a rather complex model that i have painted for printing in 4 color. But when im slicing the model i have some small places where it will print with filament 1, which is wrong.

I have attached 3 files, one image where it's painted, one image where it's sliced and a small video

Painted:
[image]

Sliced:
[image]

Video:
https://drive.google.com/open?id=1SkvW94Fg3TVyxc_lfidHP6fW6vHKUi4t

Best Regards
Bjarne Pedersen
Denmark

derekVogt, 2019-07-15:
Hi Bjarne,

Thank you for opening this issue. This is a known bug with the slicing engine currently - specifically that "Gap fill" sometimes uses the wrong extruder. Until we can properly close this issue, please consider turning off gap fill. 

Thank you,
Derek

maximuss, 2019-07-15:
Hi Derek

Thank you for response.
I have now tryed to remove "Gap fill", there is no more artifacts, thats true - but the result is even more bad. 
I will stop to try to get any good result until this issue have been solved - do you have any time line for that?

derekVogt, 2019-07-15:
Hi Bjarne,

I am surprised that the results look significantly worse with gap fill turned off. Would you be able to share the project by clicking on the share icon in the top right and either posting it here or emailing the link to me at derek (at) mosaicmfg.com.

maximuss, 2019-07-15:
ohh, its bad, allright :)
Here is the link:
https://share.canvas3d.io/projects/4ba758a5833e030c338a1bac8e81a95b

Bjarne

derekVogt, 2019-07-16:
Hi Bjarne,

I took a look through the model and have some suggestions to improve print quality. After my adjustments I am still seeing a few issues in the hinges, but I suspect that they would still print relatively well. 

To start, I would like to add is that the slicer tries its best to replicate the model exactly as it was painted. I switched one of the browns to a more contrasting color and I found a couple areas that could be cleaned up which should result in a better print. Although I do not know what your creative vision for the piece is, to the best of my ability I've fixed a couple of the triangles. I've included a couple screenshots and a share link for the model if you'd like to use my copy. 

A couple other comments that I believe would improve print quality:
- lower shell thickness, at 4 layers of 0.3mm the algorithm struggles with the angled pieces for the hinges and it doesn't allow it to ever go fully black through the inside - for this model it seems that 2 layers (0.6mm thickness) would be ideal for the hinges. 
- lower layer height, extrusion width and fewer perimeters would likely help as well

Before - painted
[image]
Before - sliced
[image]
After - painted
[image]
After - sliced
[image]

Share link: https://share.canvas3d.io/projects/53e6ccfa392c152d202eea9e459e00cb

Let me know if this helps you get closer to the result you were looking for, otherwise I might be able to further diagnose the issues.

Best,
Derek