MosaicManufacturing/canvas-issues #110: When slicing, an incorrect sized skirt/outline is created

zubairhamed, 2019-07-10:
**Describe the bug**
When slicing, an incorrect sized skirt/outline is created

**To Reproduce**
Steps to reproduce the behavior:
1. Import a model
2. Scale model
3. Slice

**Expected behavior**
The skirt/outline drawn should be around the model of the appropriate scaled size. What i'm seeing is a skirt/outline of the original imported size of the model.

**Screenshots**
[image]

**Operating System (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

zubairhamed, 2019-07-11:
also it seems that the top object (the Hono logo) bleeds into the bottom white base somehow.

brandonbloch, 2019-07-18:
Hi Zubair,

I'm pretty confident this issue is not related to the scale of the model, as scale and other transformations are applied before the files even get to the slicing engine. Rather, I believe your model is not flat on the bottom. The skirt outlines the paths of the first layer, rather than the overall shape of the model. Can you take a look at the first layer using the sliders on the right, and see what the first layer looks like?

closed 2019-08-20 by brandonbloch