MosaicManufacturing/canvas-issues #111: Changes to Project Settings not reflected in sliced GCode / Changing of Style Profile in Project Settings always pulls up the previous style settings

bash-t, 2019-07-24:
**Describe the bug**
In canvas3d.io when having multiple Project Settings (imported from a S3D Profile) the changes to the temperature done in the Project settings will not be reflected in the sliced GCode.

In addition to that, changing the "Style Profiles" within the Project Settings pulls up wrong configurations that were displayed below. It turns out that when changing the Style Profile it always shows the settings of the previously selected Style Profile.

**To Reproduce**
Steps to reproduce the behavior:
01. Go to https://prusa3d.com/downloads/Cura_S3D/S3D_OriginalPrusa_Profiles.zip to download the S3D Settings for the Prusa MK3S

02. Import the S3D Settings as Style Profiles (Original Prusa i3 MK3_MK3S.fff from the zip file)

03. It will provide you 4 profiles so that we can play around in the next steps:
 - 0.10mm DETAIL
 - 0.15mm OPTIMAL
 - 0.20mm NORMAL
 - 0.30mm FAST
 
04. Create a new Project (I made the Keychain)

05. Under "Settings" select the Style Profile "0.20mm NORMAL"

06. Then switch the Style Profile to "0.10mm DETAIL"

07. Expected behaviour: The Settings from "0.1mm DETAIL" will be displayed below.
    But instead the Settings from Step 5 "0.20mm NORMAL" is shown (this can be identified by the layer height: 0.2mm)
	
08. Switch back the Stype Profile to "0.20mm NORMAL"

09. Expected behaviour: The Settings from "0.2mm NORMAL" will be displayed below.
    But instead the Settings from Step 7 "0.10mm DETAIL" is shown (this can be identified by the layer height: 0.1mm)
	
So the Settings shown below always show the Settings _before_ you change the style profile and not the actual profile.

10. If I now want to print with the "0.2mm NORMAL" Profile, I have to select it, and then switch back to another profile, so the correct are shown below (See Screenshots)

11. But then, if I change the temperature (from 255 to 230) for example and I slice the GCode, the layer height is correct (0.2mm), but the Temperature remains at 255. (See Screenshot)

12. Expected behavior: The temperature that is set in the Project Settings is used.

**Expected behavior**
Described in Step 9 and 12

**Screenshots**
[image]
[image]

From S3D File:
[image]

[image]

[image]

**Operating System (please complete the following information):**
 - OS: Windows 10 64bit
 - Browser chrome
 - Version 75.0.3770.142

**Additional context**
I also tried using the Material Overrides to set the correct temperature, but it also not affect the sliced GCode.

brandonbloch, 2019-07-24:
Hi @bash-t, thanks for reporting! We identified and fixed this issue yesterday, and the fix will be going out later today along with a few other. I'll comment and close this issue once the update is live.

Cheers,
Brandon

bash-t, 2019-07-24:
Hi @brandonbloch thank you very much for you quick reply! I'm looking forward for testing it out once the fix is live.

Best regards,
Basti

brandonbloch, 2019-07-24:
Glad to hear! The fix is up; please re-open the issue if you have any other problems related to this.

closed 2019-07-24 by brandonbloch

bash-t, 2019-07-26:
Hi @brandonbloch 
I tested yesterday by creating/slicing another project. I can confirm that changing the "Style Profile" in the Project Settings now shows the correct settings in the sections below.

I also confirm that the given layer height is translated correctly into the sliced GCode.

However I noticed still two issues:
1. The temperature for the first layer is still 255°C, even if I changed it in the Project Settings. Also changing the Style Profile upfront in the "Printer Profiles" 

[image]

2. I activated Z-lift = 0.4 but I didn't notice any lifting of the axis during the print.
Analyzing the GCode also shows me that the Z-Axix commands only increment in 0.2mm steps (so the layers itself).

Attached you'll find the sliced GCode in case you want it to analyse [Upper-Body-agineo (2).zip](https://github.com/MosaicManufacturing/canvas-issues/files/3434809/Upper-Body-agineo.2.zip)
These are the project settings:
[image]

Having Z-Lift working would be a crucial point for me to get a crisp first layer without dragging one color into the other (like here):
[image]

Thank you very much and have a nice weekend!
Basti

brandonbloch, 2019-07-26:
Hi again,

Issue 1 – we identified a bug with first layer print temperature being forced to 255 °C when the material profile being used attempted to override the setting; is that the case for your project? This fix will be going out later today and should hopefully resolve this problem.

Issue 2 – this is one we haven't seen before (I print exclusively with Z-lifts); would you be able to share your project (including printer/style/material profiles) either here or to brandon@mosaicmfg.com so we can dig further into what's preventing the Z-lifts?

Cheers,
Brandon

bash-t, 2019-07-26:
Hi Brandon,

issue 1 - yes, I had an material override for the first layer temperature. Thanks, this fix then will solve this issue

issue 2: Just send you a link to the project

Many thanks!
Basti

brandonbloch, 2019-07-26:
Commenting here for visibility—this update is live, and seems to have fixed issue 2 as well!