MosaicManufacturing/canvas-issues #112: Purge block is partially overlapped on the model

golden-snake, 2019-07-27:
**Describe the bug**
Canvas doesn't automatically detect a purge block is too close a model before slicing (see attachment).

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Operating System (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

uilleann, 2019-07-27:
I have this as well.  Prior to the splice it looks like adequate space between the block an the print and then after splice it looks like this.  So I move the part well away from the block before splicing.  But this is definitely an issue that needs to be fixed.

golden-snake, 2019-07-27:
Chroma doesn't have that problem.

brandonbloch, 2019-07-30:
Hi there,

Yes, this is definitely an issue. The tower preview that is displayed before slicing should always constitute a worst-case size (so that what is created is always smaller than, or the same size as, the preview). However, the bug is really the slicing engine producing more transitions within a layer than is expected, hence the tower being larger than expected.

brandonbloch, 2019-11-12:
Hi all, these pesky issues should now be resolved! Happy printing!

closed 2019-11-12 by brandonbloch