MosaicManufacturing/canvas-issues #113: Emit multiple, extrusion-only gcode lines for an in-place side transition

jschuh, 2019-08-17:
**Is your feature request related to a problem? Please describe.**
The current implementation of in-place side transitions emits a single gcode extrusion command. I have had this repeatedly cause failed prints in two scenarios when used in conjunction with "Enable Print Speed Control During Splices" in the Canvas Plugin:
1. If the printer is at normal feed rate and a splice is made during the transition, the printer cannot process the M220 command to reduce its feed rate, and may exceed the Palette's ability to produce filament (causing missed pings and ensuing print failure).
2. If the Palette is currently splicing as the printer enters the transition, then it cannot process the M220 to increase its feed rate until the transition is complete; for long extrusions this may trigger the serial port timeout, resulting in an aborted print.

**Describe the solution you'd like**
Emit multiple extrusion-only gcode lines of shorter duration instead of a single line for a large extrusion. I tested two different approaches to implementing this with simple post-processing scripts: 1) divide up the in-place transition line into multiple lines and 2) stripp the X and Y movements from the along-edge multi-line output. Either approach worked equally well and provided ample opportunity for the printer to respond to the M220 command during the transition.

**Describe alternatives you've considered**
Transitioning along the edge works fine, but the extra movement is unnecessary wear on the printer and wastes a good opportunity to inspect the print while the bed would otherwise be stationary.

**Additional context**
Not being familiar with your code I don't want to assume anything, but it seems like this could be trivially implemented with the existing code path for along-edge transitions by simply not emitting the X and Y coordinates for the in-place mode.

brandonbloch, 2019-08-20:
This makes perfect sense. I think the best solution is just to split up the single purge command for "purge in place" side transitions, since the only real reason not to is marginally larger file sizes. Will investigate whether it makes more sense to do this by extrusion length (i.e. X mm at a time) or by time taken (i.e. X milliseconds at a time). If there's something I'm missing and you think one way makes more sense than the other, let us know!

Cheers,
Brandon

jschuh, 2019-08-22:
Honestly, whatever works. For my post-processing script it was easiest to just strip out the Y-axis movement, but like I mentioned above I don't know how your code is structured so I can't really say.

brandonbloch, 2019-08-29:
This has been implemented for in-place side transitions. The extrusion command is now split up, with each being limited to 10 mm of extrusion. This should be pretty safe in terms of Palette buffer size and serial timeouts, but please let us know if this needs any further tweaking!

closed 2019-08-29 by brandonbloch

jschuh, 2019-08-30:
Awesome. Thanks.