MosaicManufacturing/canvas-issues #115: Internal server error (2020) on slicing

Spooner, 2019-09-02:
**Describe the bug**
Internal server error (2020) on slicing

**To Reproduce**
Load attached stl and try to slice it.

**Expected behavior**
I expect it to be sliced.

**Operating System (please complete the following information):**
 - Ubuntu chrome

**Additional context**
STL that causes problem: [Coat Hook.zip](https://github.com/MosaicManufacturing/canvas-issues/files/3567062/Coat.Hook.zip)

brandonbloch, 2019-09-30:
@Spooner,

Thanks for providing the problematic STL! I can confirm that the issue was related to this model having two extraneous bytes at the end of the file that aren't associated with any geometry data. Our error checking logic now accounts for this and ignores the extraneous bytes, and this file slices correctly.

I'm closing the issue since the reported bug has been fixed. Out of curiousity though, how did you generate this STL? This is the first time I've come across a mal-sized STL file, and the fact that the text version is having a similar problem is intriguing—as this fix only applies to binary STLs, would you mind sharing the text version so we can look into a similar fix?

Cheers,
Brandon

closed 2019-09-30 by brandonbloch

Spooner, 2019-09-30:
Thanks for sorting that out for me.
I believe it was created by Solidworks - that client now sends .stdprt files instead and I convert them in Fusion360.
I believe the text version of that slt is [PLASTIC BODY.stl.zip](https://github.com/MosaicManufacturing/canvas-issues/files/3673530/PLASTIC.BODY.stl.zip) though it now seems to work in CANVAS, so no problem there (maybe it will explain what those extraneous bytes are about?)