MosaicManufacturing/canvas-issues #116: Can't access slice button on tablet

Spooner, 2019-09-02:
**Describe the bug**
I cannot see the slice button when using CANVAS on my tablet (Pixel C). It used to lie just below the bottom of the page and I could drag it up, but I can't get at it any more that way.

**To Reproduce**
Use CANVAS on a tablet

**Expected behavior**
Ideally, the button would always be visible at the bottom of the screen, but it was fine when I could drag the screen up to get to the button

**Screenshots**
Both in landscape and portrait, the buttons are not visible:

[image]
[image]

**Operating System (please complete the following information):**
 - Android 8.1 - Chrome

Spooner, 2019-09-30:
BUMP - this is a showstopper issue for me, but has been ignored for a month @brandonbloch

brandonbloch, 2019-09-30:
Hi @Spooner,

Sincere apologies for missing this for so long; TLDR is that my GitHub account uses my personal email address, and I spent Sunday migrating mail providers after finding out that I've been missing most of my emails for over a month. So thank you for bumping the issue!

I attempted to reproduce this issue but our Android tablet for testing doesn't seem to have a high enough resolution to do so (it's running the mobile UI, as opposed to yours which is large enough to display the full UI). I just want to confirm the screen's resolution—[this link](https://support.google.com/pixel/answer/6328677?hl=en) specifies 2560x1800 pixels for the Pixel C but please let me know if that's inaccurate.

We're going to continue digging into this and will keep you updated! Hopefully this can be resolved within the next few days. Again, we apologize for the frustration and for the delay in responding to this issue.

Cheers,
Brandon

Spooner, 2019-09-30:
Yes, that sounds like the correct size for the Pixel C (edit: The images are exactly that size, so yes that is right!).

Spooner, 2019-09-30:
And, of course, I can still use CANVAS on my PC, but sometimes it is nice to just use a tablet through the thingiverse->canvas->octoprint process :)

brandonbloch, 2019-10-02:
Hi again,

We were unfortunately unable to reproduce this issue, but we took the opportunity to clean up some of the height-related CSS in the hopes of resolving this. Nothing has broken in our testing but I'm hoping that it helped your issue! Please let us know if it did or not (if it didn't fix the issue, but something did change visually, please include an updated screenshot).

Thanks!
Brandon

Spooner, 2019-10-02:
Sorry, it looks exactly the same. The options menu is always perfectly stuck to the bottom though.

Are you using percentage viewport (vh) measurements to place it, since they can use the size of the screen rather than the size of the actual viewport? As I said, those buttons have always been just under what I can see, but I used to be able to scroll down to that position. Still, a "correct" solution would be fixing the placement rather than re-enabling vertical scrolling.

Spooner, 2019-10-02:
Just had an idea of how to workaround it and it works...split screen mode makes the window narrow enough to display properly!

[image]

brandonbloch, 2019-10-07:
Glad you found a workaround!

However, I believe this issue is finally (actually!) resolved. We procured a device that could replicate the bug and we were able to tweak the CSS to fix the issue. As such I'm closing the issue, but please re-open if there are still any problems on your end! (You may need to force-refresh or clear cache to see the change.)

Cheers,
Brandon

closed 2019-10-07 by brandonbloch

Spooner, 2019-10-07:
@brandonbloch Yeah, working fine for me too. Thanks again!

brandonbloch, 2019-10-07:
Glad to hear! Happy slicing!