MosaicManufacturing/canvas-issues #123: Add logic( "if", "then") statements

Phoenix54, 2019-11-13:
"If" and "then" statements can give you more power on how you can control your printer.
Ex. "If z<15, goto Z15; conditional statement for moving up to 15mm if less than 15mm for clearance for accessories. 
       "If z>15, goto +z0.6(as z hop);

mpkottawa, 2019-11-14:
i like this idea as i have a purge mechanism that interferes with the bed at lower z heights.

brandonbloch, 2019-11-25:
Hi all,

Thank you for the feature request. This is now being worked on, and I think you'll like what we come up with!

Cheers,
Brandon

brandonbloch, 2020-05-14:
Hi all, I know it's been a long time coming but I'm glad to finally say, this and much more is now possible! We've just released support for a new scripting language called PrinterScript that allows you to do all kinds of complex stuff within your G-code sequences. Take a look at [the docs](https://canvas3d.io/printerscript) to learn more, but here's how you might accomplish your example above:

```
minClearanceZ = 15
if (currentZ < minClearanceZ) {
  // leave clearance for accessories
  "G1 Z{{minClearanceZ}} F{{zTravelSpeed * 60}}"
} else if (currentZ > minClearanceZ) {
  // just do a Z hop
  "G1 Z{{currentZ + zLiftDistance}} F{{zTravelSpeed * 60}}"
}
```

As always, let us know if you have any questions or issues!

Cheers,
Brandon

closed 2020-05-14 by brandonbloch