MosaicManufacturing/canvas-issues #125: Two new material override features

mrac1, 2019-11-26:
In Canvas I'm missing two material overrides
- linear advance K-Factor
- bed temp

Currently, you have to create a printer for each filament with different Linear Advance settings. It would be useful to be able to assign these values to the material. Thank you in advance!

brandonbloch, 2019-12-02:
Hi @mrac1,

Thanks for the request! K-factor is pretty straightforward, but I'm unsure how per-material bed temperatures would work if you have more than one temperature for the print (this is why it was left out of the material overrides). The current set of material overrides are ones that affect that material independently (e.g. extruder temperature) rather than the entire print globally (e.g. bed temperature).

Could you please elaborate on how you'd expect a per-material bed temperature to work? E.g. should it use the hottest one in the print, the coolest, or an average? Or a different approach entirely?

Cheers,
Brandon

mrac1, 2019-12-02:
Hi @brandonbloch,

thanks for the quick response! You're right, the bed temperature I can already save in the style profile.
Filament specific GCode for M900 (Marlin) or M572 (Duet) would be great!

Kind regards,
Michael

fluffy-critter, 2020-04-01:
I can see two cases for multi-material bed temperature:

1. The lowest layers are all the same material, in which case, use that material's temperature
2. Set the temperature based either on the maximum needed for all materials (usually the correct answer), or use the weighted average of the bottom layers' materials' area (which is really just a more specific version of 1, really)

Another possibility: if the lowest layers have multiple materials, ask the user what they want to do.

Too many of my prints have failed because I forgot to verify the bed temperature for my prints (I usually print PLA but sometimes I print PETG which likes 85º on my ultrabase) and it'd be nice if I could have an additional safety net.

brandonbloch, 2020-05-14:
Hi all, just following up as we've finally been able to introduce some new G-code scripting functionality that we've been working on for some time. Part of this is the availability of a "material change" sequence, which is defined on a material profile and runs whenever switching to that material (including after your start sequence, if it's the first material used). This is now a super-simple way to associate commands with different materials, linear advance settings being a great example!

I understand now about per-material bed temperatures, I'll add it to our list but we'll need to figure out an approach that balances safety and correctness, i.e. being careful how we disambiguate. I'll close out this issue and reference it from a new one for that.

Cheers,
Brandon

closed 2020-05-14 by brandonbloch