MosaicManufacturing/canvas-issues #126: Skirt is not working with models less than 4 colours

NZSPY, 2019-11-29:
**Describe the bug**
Place multi colour models in Canvas, the has less than 3 colours, have skirt option turned on.
When sliced it shows the 4th colour as being the skirt. When trying to print model, Palette 2 pro does not load the 4th colour and fails to produce the output filament. process starts but does just sits and does nothing

**To Reproduce**
Steps to reproduce the behavior:
1. import 3 colour model
2. set the colours on the model
3. set skirt in the slice settings
4. slice model
5. Send model to Canvas hub
6. print model
7. Palette2 Asks for the 3 colours to be loaded
8. load colours
9. Splicing starts
10. Palette says a colour has run out
11. print fails

alternative steps

1. import 3 colour model
2. set the colours on the model
3. set skirt in the slice settings
4. slice model
5. Send model to Canvas hub
6. Load 4 colours into Palette2
7. Start print
8. Splicing starts
10. Palette just stops and does nothing

**Expected behavior**

Model should be sliced with the skirt using first colour to be printed

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Operating System (please complete the following information):**
 - OS: Mac OS 10.15
 - Browser Safari
 

**Additional context**

when print is canceled, Palette2 pro screen show message about compelling tasks but never does

brandonbloch, 2019-11-29:
HI @NZSPY,

Thank you for the report! Unfortunately I'm unable to reproduce the issue with the steps you've given above. Some more information that would help us investigate:

- are you using variable transition lengths?
- are any of the models painted, or simply assigned colors through drag-and-drop?

As this issue does not occur in all cases, would you be willing to share your project with us? This will help us investigate and fix the issue much more quickly. If you're not comfortable posting the link publicly you can send it to canvas@mosaicmfg.com and reference this issue.

Cheers,
Brandon

NZSPY, 2019-11-29:
> As this issue does not occur in all cases, would you be willing to share your project with us? This will help us investigate and fix the issue much more quickly. If you're not comfortable posting the link publicly you can send it to canvas@mosaicmfg.com <mailto:canvas@mosaicmfg.com> and reference this issue.

Hi Brandon, 

This is only just started to happen since your last update. I did 3 projects this morning and had the same problem with all of them, I had to remove skirt to get them to print successfully. All models are multi part with colours assigned not a painted single model

I made a copy of one to share with you

https://share.canvas3d.io/projects/c3da18861e38a5c8f543b04ae0fdb276 <https://share.canvas3d.io/projects/c3da18861e38a5c8f543b04ae0fdb276>

I have attached a screen shot from the slice page, you can see that it assigned drive 4 for the skirt. But I never used drive 4 in the model, and it then stuffs up when trying to make the first splices

Cheers
Simon

brandonbloch, 2019-12-02:
Hi Simon,

Sorry about the delay in response. Thank you for sharing the project, as it's helped us pinpoint exactly where the problem is. We're digging into it and hope to have a fix available by the end of the day.

As it turns out, the bug seems to be related to CANVAS trying to use your assigned support extruder for the brims (even though they're disabled for this print, they are set to use extruder 4). This is obviously not correct, but it at least means that as a workaround you can enable supports, change the support extruder to input 1, and disable them again, and skirts should then print with the correct extruder.

Will update here with any progress.

Cheers,
Brandon

brandonbloch, 2019-12-02:
Hi again,

We've resolved this issue and deployed the fix. Skirts/brims should now always associate with the first tool used in the print, and the workaround I mentioned above is unnecessary.

Please re-open this issue if you notice any more discrepancies!

Cheers,
Brandon

closed 2019-12-02 by brandonbloch

NZSPY, 2019-12-02:
Thanks Brandon :-)

Regards
Simon Young 
021 912433