MosaicManufacturing/canvas-issues #128: [Feature request] Arched purge block for delta with round build plate.

joris57, 2019-12-06:
Deltas in general have a relatively small build plate and a rectangular purge block takes up a disproportional share of the circular build plate.

It would be far more efficient in this case to be able to have the purge tower in an arc along the edge of the build plate, allowing either to specify the arc (e.g. between 90° and 360°) or the width of the purge tower (e.g. 10mm wide).

brandonbloch, 2019-12-12:
Hi @joris57,

Thanks for the feature request! This sounds interesting and potentially doable, but I want to make sure we understand exactly what you're hoping for. I picture functionality like this:

[image]

Could you provide a sketch of the two approaches you describe?

Cheers,
Brandon

joris57, 2019-12-14:
[image]

Hi @brandonbloch , thanks for your reply.

I was more thinking of something along the lines of this drawing. As the surface of the tower is determined by the transition parameters, the user would be able to either define the width (15 mm in this drawing) or the angle (90° in this drawing).

But I think I like your suggested approach better ... Clever.

Joris.

brandonbloch, 2019-12-16:
Ah, interesting! I see some benefits to both approaches, and we'll consider this going forward (approaches to transition tower path generation are being revisited in the new year). I'll update this thread when we get there!

Cheers,
Brandon

PS. out of curiosity, does your printer support arc move (G2/G3) commands?

joris57, 2019-12-16:
@brandonbloch I currently have Klipper installed on my deltas. Klipper does indeed support G2/G3 (and so does Marlin).