MosaicManufacturing/canvas-issues #130: [Feature request] Allow to suppress purge tower

joris57, 2019-12-16:
In cases where it is not a problem for the transition material to appear in the print, it would be nice to set be able to set the transition length to zero, effectively eliminating the purge tower.

brandonbloch, 2019-12-16:
Hi @joris57,

This is a feature that's been requested a number of times, and is actually present in our Chroma software. However the feature is very error-prone and causes much more confusion for users than the benefit it provides, which is why we made the conscious decision to exclude it from CANVAS. For this reason I'm closing the issue as "won't fix".

The long and short of it is, unfortunately there are a few technical limitations that prevent us from entirely eliminating purge sequences of some kind:

1. In accessory mode, [ping sequences](http://mm3d.co/ping-pong) contain a required stretch of extrusion that is 20 mm long. Transition towers (or side transitions) must be able to accommodate this, as we can't do the pause-extrude-pause sequence on your print without quality suffering. This isn't a requirement in connected mode (where pings are done by serial communication and are instant), but because connected pings require flushing your printer's command buffer, they can still negatively affect print quality and so we currently also require them to be done on the transition tower or in a side transition sequence.

2. This is the more far-reaching issue. Palette has a physical lower bound for the length of a splice (~8 cm), which is approximately the distance from the cutter wheel to the outgoing drive gear. When a splice would be too short (because the transition length + the amount of filament between transitions is less than this 8 cm), the transition must be lengthened to compensate and ensure Palette can create the splice. With no transition tower or side transitions, there is nowhere in the print to safely add extrusion, and because we know the result will be unproduceable by the hardware we have no choice but to error out and fail to slice. While certain models can be designed around this, it's very unclear to the user and extremely hard to design and/or paint around this limitation.

This is not to say it's a bad idea—we're committed to reducing transition waste, both in terms of material and time used, and so we also keep coming back to this topic. I hope my explanation shows that while great for certain use cases, implementing this feature in a reliable and fairly usable way is a fair bit more complicated and would require some creative logic.

Cheers,
Brandon

closed 2019-12-16 by brandonbloch

joris57, 2019-12-16:
@brandonbloch Thanks for explaining in such great detail.