MosaicManufacturing/canvas-issues #132: 'Minimal density' setting gives not the expected result.

joris57, 2019-12-22:
The image below shows a single layer of a multi-color print (2 stacked cubes) and the corresponding purge tower. As only one color is printed at any time, I would expect the purge tower to be printed at minimal infill density.

Both the 'minimal density' (of the purge tower) and the 'infill density' of the object were set at 5%. I would therefore expect a similar amount of infill when comparing the object to the purge tower. This is obviously not the case.

Also there is NO difference between a purge tower with 5% minimal density (1st image) and 10% minimal density (2nd image)

 
[image]

[image]

brandonbloch, 2020-01-02:
Hi @joris57,

This is definitely confusing but is not actually a bug. As mentioned in [this discussion](https://github.com/MosaicManufacturing/canvas-issues/issues/130#issuecomment-566253825), sparse tower layers have a minimum of 20 mm extrusion, which is currently a hard requirement for accessory mode. At your minimum density setting, the tower layer would have much less than this amount, which is why the density needs to be raised. Please note that removing this limitation for connected mode prints is on our to-do list, but will require us to validate that print calibration is unaffected by the change.

You can see this by using the same tower settings but causing the tower to be bigger (via more colors on a single layer). This print goes from 1 color to 2:

[image]

While this print goes from 1 color to 3:

[image]

You can see that the tower has become physically larger, but the density of sparse layers has decreased accordingly. This tower has a 5% minimum density but in neither case can it actually go that low.

I'm leaving this issue open as this is a change we'd like to make; however, since it's currently the intended behaviour I'll mark it as an enhancement rather than a bug.

Cheers,
Brandon

brandonbloch, 2020-02-10:
Hi again,

Thank you for your patience! This behaviour has been changed when running in connected mode—there is now no minimum extrusion for tower layers, which means that layers without any transitions will always respect your minimum density setting. The change is live, and so I'm closing this issue. Please let us know if you experience any issues with the change!

Cheers,
Brandon

closed 2020-02-10 by brandonbloch