MosaicManufacturing/canvas-issues #135: Splice settings not saving

palvarez19, 2019-12-30:
**Describe the bug**
When trying to save splice settings in CANVAS, I cannot confirm that it took after saving.

**To Reproduce**

1. Log into CANVAS
2. Go to Material Profiles
3. Create new material
4. Click on "Splice Settings"
5. Enter settings, then save
6. Click on "Splice Settings"
7.  Recently entered settings are not present

OR

1. Go to projects, select a multimaterial project
2. Select filament color
3. Select material (not default, but created one)
4. Click on splice settings
5. Enter settings, then save
6. Click on "Splice Settings"
7.  Recently entered settings are not present

**Expected behavior**
All saved splice settings should be displayed when viewing them in material profiles or within project.

**Operating System (please complete the following information):**
 - OS: Windows 10
 - Browser: Chrome 79.0.3945.88, FireFox 71.0

palvarez19, 2019-12-31:
Screenshots of where I am having the issue.
[image]

[image]

palvarez19, 2019-12-31:
Sorry, hit the wrong button and closed the issue prematurely.

leemun1, 2019-12-31:
Hello @palvarez19, I am sorry you are having this issue. Just to clarify, could you explain what you mean by "Recently entered settings are not present"? Do the values in the inputs get reset to 0 (i.e., the default value)?

If possible, would you be able to also take a screenshot when you go back into the view after attempting to save the values?

palvarez19, 2019-12-31:
@leemun1 , the values are blank.  See screenshots below of what happens after saving settings.

[image]

[image]

leemun1, 2019-12-31:
@palvarez19 in the screenshot above, the values are not showing because you have not yet selected material(s) yet. Could you select the material(s) that you set the splice settings for via the dropdown menu in the middle and confirm again please?

palvarez19, 2019-12-31:
@leemun1, you asked me to show you what it looked like after it saved.  The screenshots show what it looks like AFTER I save the settings then go back to see if they took.

brandonbloch, 2020-01-02:
Hi @palvarez19,

Sorry you're running into this! @leemun1 is right though: splice settings _are_ working correctly. You need to select the appropriate Palette type and material combination for the form to populate, which I can see you've done from the first set of screenshots. You need to do this anytime you want to edit the splice settings for that combination, including if you leave this form and come back. In your second set of screenshots you haven't done this, which is why the fields are blank.

There may be room for UX improvement here, such as remembering the last combination you were editing, but splice settings are indeed saving. As such I'm closing the issue.

Cheers,
Brandon

closed 2020-01-02 by brandonbloch