MosaicManufacturing/canvas-issues #138: Slicing keeps failing

onxygit, 2020-02-24:
Not sure why, but slicing this model (originally from thingiverse (https://www.thingiverse.com/thing:3527243) continually fails. There's no debug info shared. Could it be my profile? Settings? The model itself?

https://share.canvas3d.io/projects/49d0696b4637af4a1572017e98a940d6

brandonbloch, 2020-02-24:
Hi there,

Sorry you're running into this issue! Upon further investigation, both of these issues are caused by the STL files themselves. They contain some non-standard formatting, which our converter is currently handling incorrectly, and the resulting partially-broken geometry can't handled correctly during slicing, auto-arrange, etc.

We're working on a fix and hope to have something released today or tomorrow. In the meantime, here are the same STLs without the formatting quirks. Uploading these ones instead of the Thingiverse files should avoid the issue while we work on a fix.

Cheers,
Brandon

brandonbloch, 2020-02-24:
[Mandala-Fixed.zip](https://github.com/MosaicManufacturing/canvas-issues/files/4245410/Mandala-Fixed.zip)

onxygit, 2020-02-25:
Wow thanks for the quick response, and for addressing my issue! :)

brandonbloch, 2020-02-26:
No problem! The fix for this bug has been released and the original models should no longer cause any problems.

Cheers,
Brandon

closed 2020-02-26 by brandonbloch

onxygit, 2020-04-13:
Hey again, so I updated my design based on the model you fixed for me (see https://canvas3d.io/slicer/04520aa9c97e8cf0b9bdd85c1ca0589a). I tried printing, and it looked like the filament was generating correctly, but that my printer made it part way (I'd say about a third) and then stopped. I tried printing the .gx file (i have a FlashForge Adventurer 3 lite) with one solid color filament, bypassing the Palette, and I got the same result, so I don't think it's a splicing problem, but it appears to be a problem with the .gx file canvas generated. I'm attaching a pic of the solid color print, the generated .gx file, and a screenshot of the flashforge slicer viewing the .gx file. Notice that there's a visible difference between the bottom 1/3 of the print and the layers above. Notice also that this happens at around layer 18,000 -- I'm not sure why it's printing so many layers, since I chose a .2mm layer height. 
[image]
[image]

[Green-2 (1).zip](https://github.com/MosaicManufacturing/canvas-issues/files/4467555/Green-2.1.zip)
[image]

onxygit, 2020-04-13:
[image]
Here's a closeup from the slicer view... it looks like there's a pronounced difference in the bottom, then an air gap, and then some spaced out layers on the top!

onxygit, 2020-04-13:
[image]
Yeah, there's only like 80 something layers in Canvas and tens of thousands read by flashprint, and Canvas just gives me this :/

brandonbloch, 2020-04-14:
Hey there, sorry about this! I extracted the G-code out of the .gx file you provided and opened it with Simplify3D's visualizer. The large number of layers I'm not so sure about, but it looks to be a UI/parsing issue rather than a G-code level one as S3D lists a much more reasonable 128 layers (text box at bottom right):

[image]

As for the gaps (as well as the very strange extrusion in the print), it appears that the tool I used to fix the STLs affected the relative alignment of the models, and they need some manual adjustment aside from just aligning + grouping them. Really sorry about that, I was sure my tool maintained their original positions. You can see the gaps, and floating bodies from the mis-alignment, more clearly here:

[image]

In general, scrolling through the layers in CANVAS' visualizer can help a lot to spot issues with strange extrusion, especially due to mesh issues resulting in solid internal regions:

[image]

Since the issue with the original models has been fixed, I would recommend using them instead of my modified ones—this is generally easier than accurately aligning all the models in X/Y/Z by hand. That said, this model in particular is especially problematic and CANVAS does not currently do any built-in mesh repair or analysis, so it may be worth investigating improving these STLs. Mesh analysis/repair are on our radar but please feel free to open a new issue if you'd like to formally request it.

Finally, Chroma only opens and previews multi-extruder files, such as multi-process .gx files generated by Simplify3D, but not the single-extruder G-code CANVAS generates.

I hope that helps! Please let us know if you have any further issues or questions.

Cheers,
Brandon