MosaicManufacturing/canvas-issues #140: Stamps on top and bottom surfaces don't have fill behind them

rcmaniac25, 2020-02-28:
**Describe the bug**
First off, I see this as a bug, not a feature. Given that stamps are a picture projected onto a surface, I'd assume the surface is the background and that the colors making up the image will be placed on them. What seems to happen is the image simply replaces the surface/background. In my case, the print preview seems to indicate I will be able to see the infill through the stamped image. I'd expect that the surface will be used as fill over/under the stamp so it has a clean surface to adhere to and won't be visible through the stamp. Note: I know the preview isn't perfect. I haven't tried printing yet.

**To Reproduce**
Steps to reproduce the behavior:
1. Add a model that has a flat top or bottom.
2. Add a stamp to the top or bottom surface.
3. Slice

**Expected behavior**
Best-effort stamp applied to a surface, without infill visible.

**Screenshots**
Expectation (with my amazing /s photoshopping skills)
[image]
Reality
[image]
Gaps with infill visible (not everything circled)
[image]

**Operating System (please complete the following information):**
 - OS: Windows
 - Browser: Firefox
 - Version: 72

**Additional context**
Add any other context about the problem here.

brandonbloch, 2020-03-02:
Hi @rcmaniac25,

You're correct, this is a bug. It's not actually caused by stamping, however, but rather by poor gap fill generation. With such fine details in stamped models, the issue just presents itself more noticeably than for non-stamped models. Know that we're working on improving the algorithm for gap fill to improve print quality, especially in situations like this.

Cheers,
Brandon

rcmaniac25, 2020-03-03:
I agree that this isn't a bug of stamping, but stamping makes it really easy to make it occur without making a complex model.