MosaicManufacturing/canvas-issues #142: Feature Request - Canvas Paint Line Smoothing

opy001, 2020-03-02:
So I've been following the Palette for a while watching the issues people have and the rate at which the issues are resolved and the hardware/software are updated.  I'm quite impressed so far and am looking at ordering one in the next couple of weeks.  I have 3 printers and already have the one selected for the palette.  This will be going on my home made AM8 style printer.  I have a Hypercube with a Diamond Full Color nozzle too and it works pretty good.  I really want to see how they compare.

Right now I use Meshmixer to split STLs, but it has limitations and issues.  Separating a concave part can cause odd issues and sometimes screw up my model unless I do it in several pieces then merge those pieces together.  Basically it can get complicated.

I'm trying out the Canvas paint feature and while it looks awesome and works pretty good, there is no "smooth boundary" feature like Meshmixer has, that I have found.  Can this be implemented?

This is Canvas:
[image]

This is Meshmixer:
[image]
[image]
[image]

?

brandonbloch, 2020-03-25:
Hi @opy001,

This isn't currently possible in CANVAS, as paint data is applied on a per-triangle basis and no triangle splitting or re-meshing is done. I'm leaving this open as a feature request for visibility, and so others can chime in with their thoughts, but unfortunately right now I can't give an estimate of if/when we will be implementing this functionality.

Cheers,
Brandon

opy001, 2020-03-25:
Yea, it would have to be remeshed to create smoothing.  I can see how that would put a lot of strain on the engine processing that.  I may just stick with Meshmixer for complicated jobs like this.  I just need to get better at Meshmixer for some of the advanced features.

But it would be a cool feature to have in the future :)