MosaicManufacturing/canvas-issues #143: Traveling all over during prints - Bug Report and/or Feature Request?

opy001, 2020-03-19:
I just got my Palette 2S Pro yesterday and have been trying to calibrate it.  I've noticed odd travels for the keychain.  It starts out printing the ring of the keychain then lifts up and retracts to go to the other end of the keychain to start printing that perimeter.  Why does it not just start at the ring?  I know I have "start at corners" selected, but there is a corner at the ring.  This is a minor thing and not much masted time.  

The thing that is wasting a LOT of print time is the infill.  It prints a line then retracts z hops to another place and draws a line or two then retracts and z hops and goes to the other side of the keychain and prints a couple of lines.  I think it did this 3 times for one infill.  This is a complete waste of print time and unnecessary retracts as it should start on one side of the print and not stop until the infill is done, at least on a simple print like the keychain.

This was sliced using Canvas and the most basic settings.  Is there anything I need to look at to change this?

I guess this would be a feature request to make the slicing more efficient?

I have 3 printers and use Simplify 3D for the 2 without a Palette 2S Pro and infill is the fastest part.

brandonbloch, 2020-03-25:
Hi @opy001,

Labeling this as a bug as this shouldn't be happening. Would you mind providing a share link for the project (including printer + materials, if applicable) either here or to canvas@mosaicmfg.com so we can investigate?

Cheers,
Brandon

opy001, 2020-03-25:
I emailed a video.  Here is the project.

https://share.canvas3d.io/projects/e295588ca16446ab7a587431e32f882a

brandonbloch, 2020-03-30:
Hi @opy001,

Thank you for sending the video! To follow up, we've been able to confirm the issue on our end after importing the project, but will need some time to fully investigate and fix the issue. In the meantime, here are some general ideas that will not fix the problem per se, but would hopefully reduce the issues related to it:

- Under Extrusion, raising the "Always retract when travelling more than" threshold (e.g. to 10 mm) may reduce the amount of retraction done between these travel movements, in a way that might be more feasible than decreasing the retraction distance entirely.
- Also under Extrusion, lowering the "Wipe distance" setting (e.g. to 5 mm) may reduce the amount of excessive movement done when these retractions do happen. For closed paths like perimeters, wipes continue around the path; with open paths like infill, however, the wipe will reverse along the path, which can be jarring at fast travel speeds.

Please let me know if either of those solutions helps! We'll update this thread with progress as available.

Cheers,
Brandon