MosaicManufacturing/canvas-issues #145: Seam randomization not working

fluffy-critter, 2020-03-30:
**Describe the bug**

On round models without any corners, the seam always ends up at the specified seam angle regardless of randomization angle

**To Reproduce**

1. Model a round object
2. Set the slicer to maximally randomize the seam angles:
3. Slice the model

**Expected behavior**

Perimeter seams should be evenly distributed around the model

**Screenshots**

Perimeter seam configuration:

[image]

Sliced preview with retraction display enabled:

[image]

**Operating System (please complete the following information):**
 - OS: macOS 10.15.3
 - Browser: Firefox 74.0

**Additional context**

Model is modeled with multiple solid parts.

brandonbloch, 2020-03-30:
Hi @fluffy-critter, thank you for the report! I was able to confirm the issue. I believe this is actually related to #146 as well. Would you mind sharing the project here or to canvas@mosaicmfg.com so we can use it as a test case while investigating/fixing the issue?

Thanks,
Brandon

fluffy-critter, 2020-03-30:
Sure, I'll share both projects with you. Thanks!

brandonbloch, 2020-03-30:
Thanks for sending! We've found the issue with this setting and are currently testing the fix:

[image]

Barring any problems uncovered by further testing, expect this fix sometime tomorrow. I'll close this issue when it's been released.

Cheers,
Brandon

fluffy-critter, 2020-03-30:
Awesome! Thanks so much.

brandonbloch, 2020-03-31:
This fix has been released! Let us know if you have any further issues (#146 will remain open as we look for a solution to that separate issue).

Cheers,
Brandon

closed 2020-03-31 by brandonbloch

fluffy-critter, 2020-03-31:
Hmm, now I'm getting "An unexpected error occurred while slicing." I've tried shift-reloading to make sure there's nothing weirdly cached on the client side, and also removed/readded my project components. Nothing appears in the Javascript console either.

brandonbloch, 2020-03-31:
Very sorry about that, we had a rare deployment glitch that was causing projects with transition towers to fail during slicing, which only appeared in the production environment (classic). Should be resolved now; please let us know if it's not!

fluffy-critter, 2020-03-31:
Ah, I know how that goes. :) Slicing seems to be working again and the seams are properly randomized. Thanks!