MosaicManufacturing/canvas-issues #146: Painted models not placing seams on corners

fluffy-critter, 2020-03-30:
**Describe the bug**

When painting a round object, with the slicer set to put seams on corners when possible, the seam is not going on the corner.

**To Reproduce**

1. Model a round object in a single piece
2. Use paint mode to apply coloration
3. Set the slicer to attempt to place perimeter seams on corners

**Expected behavior**

Perimeter seams should be on color transitions within a layer, as that becomes a "corner" in the model

**Screenshots**

Slicer perimeter seam configuration:

[image]

Resulting slicing:

[image]

Resulting (ugly) seam in print:

[image]

**Operating System (please complete the following information):**
 - OS: macOS 10.15.3
 - Browser: Firefox 74.0

brandonbloch, 2020-03-30:
Fortunately it looks like the fix for #145 indeed addresses this as well:

[image]

Thanks again for reporting these issues! Will update here when the fix goes live (likely tomorrow).

fluffy-critter, 2020-03-30:
Hmm, that screenshot appears to not be showing corner seams in use on the layers with corners, and the seam randomization still seems to be enabled.

On that note, would it be possible to have it only do random seams on the layers with no corners, and use corner seams on layers with corners?

fluffy-critter, 2020-03-30:
(that said, in my experience seam randomization is way more helpful to getting a clean print than corner seams anyway)

brandonbloch, 2020-03-30:
Ah I see what you mean about the corners; I thought I had coasting enabled but I didn't. Also worth noting that some of those retracts are from the infill paths, rather than perimeters.

It seems that corners are being chosen, but now are jittered as well. Unfortunately this won't be resolved by tomorrow, but we will continue to investigate and look for a proper solution.

The feature is supposed to do exactly that (pick the best corner available, and a fully random point otherwise) but depending on the scope of/solution to this issue we may adjust the settings to allow more control, e.g. to use corners but still fall back to a certain angle + optional randomization amount.

fluffy-critter, 2020-03-30:
Ah, makes sense.

Also, worth considering that in a painted model it'd work best to always have the seam interior to the model, rather than having it anywhere on the exterior surface, at least on painted layers. I can't think of any situation where you'd have multiple colors on a painted layer and not have at least one interior point on its slice.