MosaicManufacturing/canvas-issues #147: Different infill and extruder perimeter

fluffy-critter, 2020-03-31:
**Is your feature request related to a problem? Please describe.**

I do a lot of printing with transparent PLA and PETG, and one of the things I've really wanted to be able to do is have the perimeter be transparent while the infill is a different, opaque/colored material. In PrusaSlicer you are able to specify different extruders for perimeters, infill, and solid infill; it'd be great if Canvas could do this too.

**Describe the solution you'd like**

Just as it says above - be able to specify a filament for infill separately from the perimeter filament.

**Describe alternatives you've considered**

Obviously I could do this with PrusaSlicer+Chroma but I like having a single app that works end to end. :)

rcmaniac25, 2020-07-15:
This would be useful.

One idea that came to mind that is related to this, though more complex so may not have as much usage: filament priorities.

If I go off and have a cheap $20 spool of blue, and then I had some fancy sparkly/matte/etc. blue filament that was $40+ a spool, I don't really want to print the whole model with the $40+ spool. So there could be a feature, even for single-color models where you could say "$40+ spool is perimeters, top and bottom layers, and the $20 spool is everything not outwardly visible".

One benefit is you don't need a transition tower since you could just transition the perimeter into the (less expensive) infill and just not even transition if the visible -> internal -> visible transition can't be done in the given amount of space.

Take that a step further and you could do something like "perimeter is carbon fiber, infill is a foaming filament (ex. ColorFabb LW filament)" to be used to make lightweight prints for aviation.

But it starts with being able to assign colors to specific print elements as OP specified.

fluffy-critter, 2020-07-15:
Ooh yes I absolutely love that idea