MosaicManufacturing/canvas-issues #148: Infill not working correctly

opy001, 2020-03-31:
When infill percent is set to 0% I still get infill.  This should be hollow but in the preview it shows there are infill.  They are consistent with the type of infill I choose even though it is always set to 0%.  This shared project was sent to me by another user and I am getting the same results as him.  I am using a different printer profile so I know that is not the issue.

https://share.canvas3d.io/projects/574b46bb9ecff7ddd684491cdea63341

brandonbloch, 2020-03-31:
Hi, I can reproduce the issue on this or any other project. The issue was caused by an incorrect bound being placed on infill density, as it works correctly when vase mode is enabled. The change needed is small and easy to test, so a fix should be live shortly!

brandonbloch, 2020-03-31:
This fix is out! Confirmed with the shared project just now.

closed 2020-03-31 by brandonbloch

opy001, 2020-03-31:
You guys are awesome!  Thanks!