MosaicManufacturing/canvas-issues #150: Stretch/Skew stamps

opy001, 2020-04-04:
When placing a stamp it is great that I can rotate and resize, but stretching length or height alone would be great or skewing such as grabbing a corner and just stretching that one corner to skew it a bit would be awesome!  It would reduce the amount of stamps I have to use or amount of time I have to make them in photoshop.

leemun1, 2020-04-04:
Hello @opy001 , you can do that by holding down on the SHIFT key while you scale your image using one of the corner handles. Apologies for not making it easily discoverable.

opy001, 2020-04-04:
I may just be blind.  I saw the space bar hinT, but didn’t see the shift key hint.  I should’ve known you guys wouldn’t leave something that basic and awesome out.  Thanks!

opy001, 2020-04-04:
Ok I can stretch one way or another and that is awesome!

I'd like to skew or warp it into a quadrilateral shape.  That way I can do stuff such as make different stripes that are fatter on one side than the other or even a rhombus from a regular square.

Is that at all possible?

leemun1, 2020-04-04:
@opy001 Unfortunately, Canvas currently does not have a skewing/warping feature for stamp images. Your image needs to have those transformations applied already before you upload them to Canvas. That being said, we will add that feature to our backlog of potential future enhancements. Thank you for your suggestion!

opy001, 2020-04-04:
Thanks!  It'll help with the minimization of stamps I have to use.  I've got around 30 right now to fit the angles I need and I'm not done yet.

Is there a limit of stamps the system can handle?

leemun1, 2020-04-06:
@opy001 Although Canvas does not explicitly limit the number of stamps you can upload, you are still constrained by your machine's capabilities. Stamping leverages GPU resources, so the more images you upload (and the bigger those images are), the chances of you being left with no more GPU resources on your machine becomes higher.

If your use case requires uploading of many images, then we advise that you try to use images that are smaller in size.