MosaicManufacturing/canvas-issues #151: Zoomes to much

ShakieVan, 2020-04-09:
**Describe the bug**
With the mouse wheel it zoomes way to much in every 3d view.

The pictures show the difference between 1 wheel step.
My mouse wheel is configured at Windows standard (3 lines per wheel step)

**To Reproduce**
Steps to reproduce the behavior:
1. Open a project (3d view)
2. Zoom in with scroll wheel by one wheel step

**Expected behavior**
A smoother zoom not that huge difference between one wheel step.

**Screenshots**
[mousewheel0](/uploads/8656ca05a8c81dd9d64b7fb8fe050620/mousewheel0.jpg)!
[mousewheel1](/uploads/5a6e0f4a634baefbf2a62701be506b4d/mousewheel1.jpg)

**Operating System (please complete the following information):**
 - OS: Windows 10 (1909)
 - Firefox 74.0.1
 - Chrome 80.0.3987.163
 - Logitech MX Ergo (FW 006.003.00022)

ShakieVan, 2020-04-09:
I experienced something weird. When the focus is on another application an I scroll in Canvas (in the background) everything is OK.

But when focus is on the browser, scrolling is as described in the first post.

brandonbloch, 2020-04-27:
Hi @ShakieVan,

Our team has been unable to reproduce this issue (including one member who I believe has the same mouse). Would you be able to provide a few more details that might help?

1. Does the issue happen in any other browser, or just Firefox?
2. Could you provide a video of the scrolling one wheel step at a time when the window is focused vs. another window being on focus?

Thanks,
Brandon

accetturo, 2020-06-01:
I experience a similar issue where zooming is extremely off with the click wheel, I use a Logitech master s2