MosaicManufacturing/canvas-issues #153: Stamping on curved surface is sliced incorrectly

stwf, 2020-04-19:
**Describe the bug**
I applied a png image stamp (just of some text) to the from face of a curved surface. While it looks fine when rendered on screen the resulting print was flattened out where the stamp was.

**To Reproduce**
Here is the model:
https://share.canvas3d.io/projects/404d00325b69339afc6a95df4b376967

I am attaching a pic of the printed result.

**Expected behavior**
Expected the shape of the model to be retained.

**Screenshots**
[image]
[image]

**Operating System (please complete the following information):**
 - Mac
 - Safari, but the web side of things looked fine. I believe it was slicing issue.
 - Printed on a Monoprice Ultimate if that matters but the model has printed many times without the stamp and looks good.

brandonbloch, 2020-04-27:
Hi @stwf,

Would you mind attaching the STL file directly or sending it to canvas@mosaicmfg.com so we can take a look at it in some other software? I'd like to narrow down whether it's due to some specific feature of this geometry in particular (e.g. extra triangles in erroneous places) or a general issue such as improper handling of thin stamped regions.

Thanks,
Brandon

stwf, 2020-04-27:
[MASK.STL.zip](https://github.com/MosaicManufacturing/canvas-issues/files/4540345/MASK.STL.zip)

stwf, 2020-04-27:
github insisted on the zip