MosaicManufacturing/canvas-issues #154: Sharing project is glitchy

opy001, 2020-04-25:
While editing a project I tried to share it.  The "model" option to share was grayed out.  I went to the main projects screen and it was grayed out there as well.  I could not share any model at all, just the printer profile and style settings.  I tried it from my phone and from the main projects screen I could share models.  I then opened a new tab on my computer and on the main projects screen I could share models again.  I then opened a project and tried to share it and the model option was grayed out again so I went back to the main projects screen and some of them had the model option grayed out.  

Definitely a glitch.  My dad is having similar issues.

brandonbloch, 2020-04-27:
Hi @opy001,

Thanks for reporting! I've reproduced the issue on my end and we should be able to get a fix out this morning.

brandonbloch, 2020-04-27:
The fix has been released (you may need to hard-refresh your browser). Let us know if you still have any issues!

Cheers,
Brandon

closed 2020-04-27 by brandonbloch

opy001, 2020-04-27:
It is definitely better, but still a little odd.  I did close the browser and re-open it.

I was able to share a model and printer profile/settings within the project itself.  I left the project and went to the main projects page and selected that project and chose to share and the model option was unchecked and grayed out.  I can go back into the project and it is still selected.

I then shared a different model from the main projects page and was able to select model and printer profile/settings.  I then went into the project and it looked correct with both model and printer profile/settings selected.  Then I went back to the main projects page and selected that model and hit share and the model selection was unchecked and grayed out.  I went into the project itself and the model was still shared.

So sharing works, but the display is glitchy under certain uncommon circumstances.

Sorry, my actual job is the try to break things to see what needs to be improved, so I try things that would not normally be done all the time.