MosaicManufacturing/canvas-issues #155: Download Issues iPadOS

gingerninja93, 2020-04-28:
Previously, I had been using Canvas as my go to 3D slicer for my printers, as it allows me to use just my iPad and leave my laptop out of the picture. This was working great for months, just recently I have been unable to complete any downloads through safari, chrome, or Firefox.

Currently, when I hit “download” after slicing the model, I get the message “Safari cannot open the  page, the operation couldn’t be completed, Webkitblobresource error 1”.

[image]

Previously, the file would pop up and I could choose a download location. 

Currently experiencing the issue on both iPadOS and iOS. 

Would love to find a fix since I do all my 3D design on the iPad and having the slicer available on the same device was a game changer.

leemun1, 2020-04-29:
@gingerninja93
Thank you for reporting this issue.
We were able to reproduce this issue on an iPad, and are now working on a fix.

leemun1, 2020-05-05:
@gingerninja93
We pushed an update that involves a fix to this issue.
This issue will now be closed.

closed 2020-05-05 by leemun1