MosaicManufacturing/canvas-issues #157: Visualiser shows drive 1 and 2 material length when only drive 2 in use

cscashby, 2020-06-09:
**Describe the bug**
When previewing sliced file, print summary shows:

Drive 1: 77.24 m
Drive 2: 5.44 m

When in fact there is no transition, and the one object is using drive 2.

**To Reproduce**
Steps to reproduce the behavior:
1. Use an old 2-material model as a template (copy the design)
2. Delete model, and drag in a new one
3. Drag material 2 to the new model (I want those specs for this design as it's a one-colour part of a larger multi-material construction)
4. Edit settings and ensure supports and infill is in material 2
5. Slice

**Expected behavior**
Slice shows one material and previews correctly

**Screenshots**
Available on request (do not want to post model publicly)

**Operating System (please complete the following information):**
 - OS: Mac OS Catalina
 - Browser: Chrome
 - Version: 83.0.4103.97 (Official Build) (64-bit)

brandonbloch, 2020-06-18:
Hi @cscashby, it sounds like something funky is definitely going on here. Would you mind [sharing your project)(http://mm3d.co/canvas-sharing) either here or to canvas@mosaicmfg.com so we can take a look? Please be sure to include the printer profile and any material profiles in case those are part of the issue.

Thanks,
Brandon

cscashby, 2020-06-18:
Hrm, damnit - I can't reproduce any more! I'll try again when I next clone a model but happy to close for now. Will reopen if I can find out what's going on. Thanks for the follow up.

brandonbloch, 2020-06-18:
No worries, feel free to re-open the issue if you encounter it again!