MosaicManufacturing/canvas-issues #160: Feature request: No sparse layers when not needed

Argolein, 2020-06-14:
Feature: No sparse layers are printed when there is no transition to reduce filament waste. 

Feature seen in Prusa Slicer in MMU mode for their MMU(2S) multi material addon and seen in the **P2PP script for Prusa Slicer and the Palette 2** with "Full Purge Tower reduction" feature enabled.

For P2PP the feature is documented here: https://github.com/tomvandeneede/p2pp/blob/dev/P2PP%20user%20manual.pdf

The P2PP dev branch (newest version):
https://github.com/tomvandeneede/p2pp/tree/dev

The feature would need a set "max purge tower delta" so the height difference between purge tower and print doesn't get too big so it can auto add purge to stay within the height difference.

brandonbloch, 2020-06-18:
Hi @Argolein, thanks for the feature request!

This is a really cool feature and we definitely see the material- and time-saving benefit in adding something like this. Unfortunately it gets a fair bit more complicated to implement in a cross-printer way—P2PP can get away with this because it's specifically designed for the Prusa i3 and has a very good understanding of the extruder and gantry geometry. Even in P2PP, it also requires a fair bit of manual work and care to ensure that you arrange things in a way that this feature won't end up ruining your print or damaging your extruder/gantry somehow.

We feel that the best solution we could build would handle all of this, having enough of an understanding of the printer's geometry to eliminate the guesswork on the user's part, and working across any printer that can physically handle this feature. This is of course a large undertaking, and there are lots of areas we're working to improve Canvas, but rest assured that waste reduction is an important goal for us and we're constantly considering solutions such as this. I can't give a timeline, but I think it's likely this feature eventually gets built in some form.

Cheers,
Brandon

rcmaniac25, 2020-07-15:
I very much want this feature as well. I forgot where I saw it, but I'm pretty sure there was some slicer that let you simply specify the extruder size (and maybe the offset for the nozzle within it). You could duplicate the Bed UI for the Extruder when editing the printer, and just leave it as unset/default when a printer first gets created. If a print mode is specified to skip sparse layers, and the extruder is unset/default, it just says "can't until extruder is set".