MosaicManufacturing/canvas-issues #163: Non-integer rotate or autorotate?

cscashby, 2020-06-18:
**Is your feature request related to a problem? Please describe.**
Currently it's not possible to rotate by a non-integer number of degrees

**Describe the solution you'd like**
Allow floats in the rotation box. Or, even better (as well?) have an option to select a face which should be rotated to the closest multiple of 90 degrees to the XY plane to align a model properly.

**Additional context**
This is important for me in trying to align models with soluble / alternate material supports to minimise the support layer and make boundaries properly flat

leemun1, 2020-06-18:
@cscashby
Thank you for submitting this issue.
We apologize for any inconvenience that this may have caused in your workflow.
While you currently can only rotate models in integer degrees, you can also "snap" to closest axes or the diagonals of each (i.e., rotate by 45 degrees in all direction) by holding down the CTRL key while using the rotation gizmo handles (attached picture below).

[image]

Hope this helps!

cscashby, 2020-06-18:
Thanks, I'm aware of this. The problem is that the face I'm wanting to straighten is not at an integer number of degrees to a x/y/z plane, so this doesn't help.

Absolutely not a big deal but a good enhancement request I think.

leemun1, 2020-06-18:
@cscashby 

Ah, I see.
A temporary workaround I could think of is to make use of the `Lay face to bed` tool in combination with the rotation snap feature.

For example, once you use the `Lay face to bed` tool to align one of the face to be parallel to the print bed, you can then apply rotations in multiples of 90.

You can find the tool under the `More` tool (attached picture below).

[image]

I apologize in advance if this does not solve anything; we will add this to our list of potential future enhancements.

cscashby, 2020-06-18:
Ah-ha! I'd seen the feature ages ago and forgot about it - apologies.