MosaicManufacturing/canvas-issues #167: after slicing its showing my project is being printed in one color when it is not

8002imas, 2020-07-02:
**Describe the bug**after slicing its showing my project is being printed in one color when it is not

Steps to reproduce the behavior:
1. Go to new project
2. color parts of the project in 2 colors in my case yellow and white
3.slice the project
4. showing it in one color

I expected it to be shown in Multiple colors like the project

**Operating System (please complete the following information):**
 - OS macOS Catalina 
 - Browser safari 
 - Version 10.15.5

leemun1, 2020-07-06:
Could you confirm that your project is using a printer profile that has a Palette attached to it?

8002imas, 2020-07-06:
I just checked and it did not I when I added it it fixed the problem.

closed 2020-07-06 by 8002imas