MosaicManufacturing/canvas-issues #175: Infill added where none should be.

cephdon, 2020-07-25:
I sliced the attached STL and didn't notice that the resulting gcode included infill where a hole should be.
In fact, this file will slice properly on Cura, but it has this weird effect on Canvas.
To help you resolve the issue, I have attached the STL, and resulting gcode file and my infill settings.

[Mask_body_L_wide.zip](https://github.com/MosaicManufacturing/canvas-issues/files/4976044/Mask_body_L_wide.zip)
[Mask-body-L-wide-2.zip](https://github.com/MosaicManufacturing/canvas-issues/files/4976045/Mask-body-L-wide-2.zip)
[image]

cephdon, 2020-07-25:
Here are the images of the correct slicing and incorrect slicing between layers 20 and 21.
[image]
[image]

brandonbloch, 2020-11-13:
Hi @cephdon, sorry about the delay here. It looks like this was caused by a mesh error (confirmed with Simplify3D's analysis tools) which proved a bit tricky to repair, but here is a repaired file that I got to slice without any issues:

[Mask_body_L_wide_fixed.stl.zip](https://github.com/MosaicManufacturing/canvas-issues/files/5539139/Mask_body_L_wide_fixed.stl.zip)

[image]

I'm closing the issue as it has a resolution. However, if you're interested in mesh analysis and/or repair as a feature in CANVAS, please feel free to open a feature request!

closed 2020-11-13 by brandonbloch