MosaicManufacturing/canvas-issues #177: Prusa MK2 No Multi material Support, no transition tower

Hopalonger, 2020-09-10:
**Describe the bug**
When I use the Prusa MK2 Profile I don't get a transition tower in the view, and I am unable to use the multi-material functions of my Palette 

**To Reproduce**
Steps to reproduce the behavior:
1: use Prusa preset
2: Change to any other preset in my case Sidewinder X1
3: See transition Tower appear
4: change back and tower disappears
5: attempt to slice and now luck

**Expected behavior**
When using this profile I expected for a transition tower to appear, and for me to be able to use the multilateral functionality

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Operating System (please complete the following information):**
 - OS: Windows 10
 - Browser Chrome
 - Version 85.0.4183.102

**Additional context**
Add any other context about the problem here.

derekVogt, 2020-09-14:
Hi Hipalonger,

Thanks for reaching out!
Would you be able to check in your Printer profiles that a Palette has been associated with your Prusa profile?
[canvas3d.io/printers/](https://canvas3d.io/printers/) (this can also be accessed via the side menu)

If there already was, let us know - almost definitely a bug on our end! 
If there wasn't, click the "Add Palette" button and select which model you have, then create a new project and you should be off to the races!

[image]

brandonbloch, 2020-11-13:
(Closing due to inactivity)

closed 2020-11-13 by brandonbloch