MosaicManufacturing/canvas-issues #180: CurrentTool / NextTool constants in PrinterScript

DataDink, 2020-09-23:
**Is your feature request related to a problem? Please describe.**
There was a thing I was trying to do, but I couldn't find a constant letting me know what tool or material was active / being changed to

**Describe the solution you'd like**
2 numeric constants added to PrinterScript describing the current / next tool during a material change sequence.

brandonbloch, 2020-11-13:
This should be a straightforward addition. Just curious from #178, would you find any value in constants for the RGBA values of the (next and/or current) tool's color as well? I'm not personally familiar with operating a color-mixing printer, but I'm wondering if you can convert RGB(A) to CMYW automatically in PrinterScript, and then you'd only need one material profile

DataDink, 2020-11-14:
Yes! that would be useful.