MosaicManufacturing/canvas-issues #183: Large File Times Out on Canvas Website.  No render completed.

Eebel, 2020-10-04:
**Describe the bug**
Canvas website will not render my x-day print.  The server times out after several minutes and the slicing never completes.

**To Reproduce**
Steps to reproduce the behavior:
1. Load a huge g.code file
2. Try to slice it on the web
3. Sit for 10 minutes while it tries
4. Watch it fail every time.

**Expected behavior**
A quick and responsive slice time that can handle large files.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Operating System (please complete the following information):**
 - OS: MacOS 10.15.7
 - Browser Safari 14

**Additional context**
I bought the Palette 2S Pro to do multi-day prints without having to worry about getting up at 3AM to change filament.  However, the website cannot handle large files.  I am printing full size Astromech droids on a 600x600x600 Modix Big60.

brandonbloch, 2020-11-13:
Hi @Eebel,

Your project is probably simply too large for our servers to handle—the limits on slice time and memory usage you're running into are in place to prevent our servers from crashing and taking down other users' slicing jobs with them. Slicer performance is one of our top priorities, and we aim to be able to support projects this large with ease, but as it is now you may need to use an offline solution in order to prepare a print this large.

Sincerely,
Brandon

Eebel, 2020-11-13:
It would seem you need to upgrade the servers to handle projects of this size and bigger.  I bought this product to do large prints with multiple spools as this feature is one of the selling points.

Tim