MosaicManufacturing/canvas-issues #189: Models outside the build area stull get sliced

rcmaniac25, 2020-11-09:
**Describe the bug**
Models outside the build area still get sliced and generated gcode and transition tower values.

**To Reproduce**
Steps to reproduce the behavior:
1. Add multiple models to a project.
2. Move one outside the printer's build area.
3. Slice the project.

**Expected behavior**
Only the models that are within the build area are sliced (this includes which filaments are used, so if one model uses filaments 1 and 2, and another outside the build area uses 1 and 3, then only the first model is sliced and the gcode/maf should only be using filaments 1 and 2)

**Additional context**
Slic3r, PrusaSlicer, Cura, and others don't slice any models outside the build area.

brandonbloch, 2020-11-13:
Hi @rcmaniac25, this is intentional, as some printers have a printable area that extends past the print bed (the reason side transitions work!), and some printer profiles (especially those coming from other slicers) have inaccurate origin positioning, so we don't make any assumptions about what is or isn't "too far" off the edges.

We may be able to add some sort of UI indication that the model may be outside the print volume, but it sounds like that doesn't solve your use case. Out of curiosity, if we were to skip such models, how should models that intersect the build volume be handled? Should they be clipped at the boundary or skipped entirely?

In the meantime, you'll have to remove the undesired model(s) from the project before slicing.

rcmaniac25, 2020-11-13:
As my most common slicer is PrusaSlicer, I'd have to answer with how it does it, which is that it skips it entirely.

This issue came up because of #186. I painted a model... then was asked to have only a subset of colors. I realized colors didn't get grouped, so I made #186. As a workaround, I duplicated the model and manually painted it, putting the version with more painted colors outside the build area... but found out that it still sliced it, leading to this issue.

As a simple alternative (again, based off my experience with PrusaSlicer): just have a toggle option to enable/disable the model slicing.

brandonbloch, 2021-01-04:
Hi @rcmaniac25,

We've decided not to try and make any assumptions about printable area vs. build plate dimensions, so we aren't clipping models or omitting those outside the build volume. Instead we figured a more direct option makes a lot of sense here—now when slicing, the show/hide toggle is used to exclude models which are currently hidden. This should solve your use case, and also allow you to maintain all the desired positioning when switching between different "sets" of models in the project.

This change is now live in the slicer; expect some accompanying front-end improvements to follow shortly.

Cheers,
Brandon

closed 2021-01-04 by brandonbloch

rcmaniac25, 2021-01-04:
Cool. I will have to try it out.