MosaicManufacturing/canvas-issues #190: Small color change-elements are printed very slowly

rcmaniac25, 2020-11-09:
**Describe the bug**
When printing items with small color changed items (such as a dot on a set of dice), the first couple layers or items that are small in general, will print extremely slowly.

**To Reproduce**
Steps to reproduce the behavior:
1. Find a print that isn't large
2. Change the color of 1 or so facets
3. Slice (should see 1-2 lines of a different color in the gcode viewer. If not, color more facets)
4. If no gocde viewer that shows print speed, then print and watch for the alt-color prints.

**Expected behavior**
Prints slow (being it's a small perimeter), but is noticeable that it's actually printing (as opposed to looking like it froze/paused printing)

**Additional context**
The first couple layers of one print I had was 1-3 perimeters width, so 0.45-1mm wide, and 1-3mm long, and this took 5-15 sec to print. It printed so slow that I thought the print had either frozen, paused, or it was doing a ping ON the print instead of the transition tower.

brandonbloch, 2020-11-13:
Hi @rcmaniac25, what is the "minimum layer time" value (under Temperature and Cooling) you're using? This setting slows down the paths on a layer to ensure it takes at least that many seconds to print. I suspect the default value may be a bit high for your printer, and that lowering this value will help. There's currently a bug which causes the transition tower to not factor into the calculation, so going as low as 0 when printing with a tower should be safe.

rcmaniac25, 2020-11-13:
Ah, I didn't see this option. Yes, mine was at a default of 20s. Perhaps an alternative/addition could be a min-print speed, so it's not scaled below that print speed.

brandonbloch, 2020-11-18:
Minimum print speed as a setting makes a lot of sense!