MosaicManufacturing/canvas-issues #197: QUESTION: Slicing

dabbodev, 2020-12-07:
Quick question because I've been using this on my laptop out of convenience but does canvas slice in-browser or in the cloud? I just want to know if slicing it on a better computer would be faster or not.

derekVogt, 2020-12-08:
Hey dabbodev,

Good question! The slicing itself runs in the cloud and is independent of your computer's specifications. That said, a better computer will improve your experience in the 3D visualizer and when painting or stamping parts. I'll also add that a fast and stable internet connection can help throughout Canvas, for example speeding up sending data to the servers or during the download of previews.

leemun1, 2020-12-11:
Closing this issue as the question has been answered.

closed 2020-12-11 by leemun1