MosaicManufacturing/canvas-issues #199: Skirt Generates Around Tower Instead Of Object

jph290, 2020-12-29:
When I generate a multi-material/multi-color print, the first material of the object prints immediately instead of having the skirt/brim print first.  The skirt is then generated around the transition tower as a brim.  I had the separation of the skirt set to 6mm and it generated a brim around the transition tower.

brandonbloch, 2021-01-04:
Hi @jph290, thanks for reaching out!

The transition tower's brim is independent of the rest of the print, and is used for adhesion as well as ensuring Palette's first splice is long enough that the device can produce it. As such, the result is always a brim (i.e. 0 offset, rather than a skirt), and while there's a setting for the tower's minimum brim count (in the Transition section) the slicer may add more than this minimum if necessary.

The main print's skirt/brim should always be the first thing printed, right after the start sequence. If you've got the setting enabled, something else could be causing it not to be generated (again, this would be independent of the transition tower). Would you be able to share your project here or to canvas@mosaicmfg.com so we can take a closer look at the issue?

Cheers,
Brandon

jph290, 2021-01-04:
Here is a link to the project.  https://share.canvas3d.io/projects/356fbf8916feea4e4868c171c294eb20

brandonbloch, 2021-01-04:
Thanks for responding so quickly!

It looks like your project doesn't have a skirt/brim enabled:

[image]

After toggling the checkbox (and setting the horizontal gap to 6 mm) the skirt appears as expected:

[image]

I'm closing this issue as resolved, but please feel free to comment/reopen if it still isn't working for you!

Cheers,
Brandon

closed 2021-01-04 by brandonbloch