MosaicManufacturing/canvas-issues #205: Failed to Fetch, unable to send files to Canvas Hub

hornetrider, 2021-02-02:
Hi
this evening I have registered the issue that it is no longer possible to send files to Canvas Hub from canvas3d.io.
Error Message. Failed to fetch

Canvas Hub: latest stable release 1.5.3.
Browser: 
Chrome Version 88.0.4324.104 (Offizieller Build) (64-Bit)
Edge Version 88.0.705.53 (Offizielles Build) (64-Bit)

Please help, as this is keeping me from doing the prints I need.

adenix, 2021-02-03:
I am also seeing this error intermittently along with others about CORS issues. They usually clear up after a few minutes.

sahanim, 2021-02-08:
Hi @hornetrider,

I'm looking into this issue now. 

Please note that you can always hit the download file button beside the 'Send to CANVAS Hub' button and use the canvas-hub (octoprint) upload file feature. 

Canvas - after slicing a project
[image]

Canvas Hub - upload file
[image]