MosaicManufacturing/canvas-issues #216: Octoprint Hub showing as offline in Canvas3D

Webfeger, 2021-05-13:
**Describe the bug**
Octoprint hub is showing as offline in Canvas3D but Canvas Plugin is working and device is linked and connected to canvas account

**To Reproduce**
Steps to reproduce the behavior:
1. Open canvased.io
2. Select Devices in left menu
3. Octoprint hub is showing as offline with orange icon

This issue occurs in Octoprint 1.5.3 and 1.6.1

mosaicmradul, 2021-06-11:
Hi @Webfeger.

The [latest Canvas Plugin build](https://gitlab.com/mosaic-mfg/canvas-plugin/-/releases/3.0.3) (3.0.3) has a 'Reset Canvas Data' button inside the Canvas Settings menu.

[image]

Resetting your canvas data will prompt you to restart OctoPrint. 

Could you please try this and let us know if this issue persists.

Webfeger, 2021-06-13:
Hello,
i did install the new plugin and follow the steps from your last post. After them i tried to add the octoprint to canvas. But it will still shown as offline:

[image]
[image]
[image]

dcsublime, 2021-08-12:
I opened an email ticket with support and am just seeing this.  I too have the same issue and have also tried the 'Reset Canvas Data" per support request.  I am still experiencing the issue as well.  Has this been resolved or anything else I can try?