MosaicManufacturing/canvas-issues #218: Cannot Import Printer Profile

BlacMac, 2021-05-22:
**Describe the bug**
Attempted to setup my Prusa i3 Mk2.5 in Canvas. When I uploaded my .ini from Prusa Slic3r, I would get the error: "Cannot read property 'split' of undefined".

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'My Setups>My Printers>Import Printer Profile'
2. Click on 'Click or drag here to upload'
3. Scroll down to '[Original Prusa i3 MK2.5 - Copy.txt](https://github.com/MosaicManufacturing/canvas-issues/files/6526845/Original.Prusa.i3.MK2.5.-.Copy.txt) Had to convert from .ini to .txt for upload'
4. See error

**Expected behavior**
Setup my printer for use. 

**Screenshots**
[image]

**Operating System (please complete the following information):**
 - OS: Windows 10
 - Browser Chrome
 - Version Chrome Version 90.0.4430.212

**Additional context**
Add any other context about the problem here.

mosaicmradul, 2021-05-22:
Hi BlacMac,

Thanks for reporting this issue. 

We’ll take a look on Tuesday after the long weekend in Canada

BlacMac, 2021-05-25:
[image]
Trying on my Mac and I am getting this error.

mosaicmradul, 2021-06-11:
Hi BlacMac,

There are several points of failure in the printer profile and it is non-salvageable.

 I am seeing that the profile is 1/4 of the size of a default Prusa i3 Mk2.5 config export.
 
 I think the best course of action here is to try and recreate a new printer profile using the latest [prusa slicer](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.1) build (2.3.1)

BlacMac, 2021-06-12:
mosaicmradul,

Thank you for pointing out the size discrepancy. I had no idea the it was different than what was expected. I updated to the latest (2.3.1) and exported my profile with no luck. I did se that we could add a printer model to the list of printers though when I started from scratch. I will try this route for now and will come back with more question if needed. Thank you!

closed 2021-06-12 by BlacMac