MosaicManufacturing/canvas-issues #232: Canvas material profile randomly swapping ingoing and outgoing profile numbers (Palette 3 Pro)

ybader, 2022-04-11:
**Describe the bug**
I seem to run into a problem where Canvas is feeding the wrong splicing profile numbers for my Palette 3 Pro. For example, I am trying to splice two filaments, white and red. Each of these have their own material profile, let's say whitepla and redpla. Using the splice tuning on the Palette, I get all the numbers and input them into canvas material profiles (yes I **quadruple** checked that I did input the correct numbers into the ingoing and outgoing profiles and that I did not mix them up). The numbers I set up for when whitepla is ingoing and redpla are outgoing are being applied to splices where redpla is ingoing and whitepla is outgoing, and vice versa. Basically the splice numbers are being applied in reverse.

**To Reproduce**
1. Make redpla in and whitepla out have extreme splice values (-3 0 -3), and set whitepla in and redpla out to the opposite extreme (3 0 3). 
2. Splice and start a print using these two filament profiles.
3. Observe splicing. What you should expect to see is that white-to-red transitions are brittle and don't splice properly, while red-to-white transitions have necking due to overheating. Instead, I end up seeing the exact reverse. I see that red-to-white are brittle and white-to-red are necking.
4. Now change the redpla in and whitepla out to (3 0 3) and re-splice and print. 
5. Observe splicing. Even though values for white-to-red were changed, now red-to-white splicing behavior has changed and is also necking.

**Operating System (please complete the following information):**
 - OS: Windows 10
 - Browser: Chrome
 - Version: Palette is running on 22.02.09-alt firmware, I tried it on the newest firmware as well but I found the new firmware extremely buggy (splice tuning is very broken on the new firmware).

**Additional context**
I haven't had a chance to fully test the cause, but I believe this bug depends on what inputs you assign your filaments to. For example, in the print that I was having the bug occur, I was running red in input 4 and white in input 5 (I was also running black in input 6). When I changed the input numbers for all the filaments, the bug seemingly stopped happening (I moved red to input 2, white to input 3, and black to input 4).
My wild guess is that this is a bug unique to the Palette 3 Pro when assigning filaments across the input 4-5 barrier. I'm guessing the software is mainly set up for the regular Palette 3 (which only has inputs 1-4), so something wacky is happening using inputs 4, 5, and 6 in my test print that has been giving me problems.

derekVogt, 2022-04-21:
Hello ybader!

Thank you for the detailed bug report! 
We have been able to reproduce the issue and will work on a fix shortly.

Specifically we've found two issues with splice tuning and material profiles:

1. The ingoing and outgoing names are swapped in Canvas. The left material should be the "Ingoing (to) material" and the right material should be the "Outgoing (from) material". Until the text is fixed, inputting your values by reversing what Canvas asks for (out->in, in->out) will translate to proper splicing values in the Palette 3 (and P+, P2) files. This appears to be a new issue caused by our recent visual refresh of Canvas.
2. We have also found that the modal displaying splice settings within a project can be confusing. Specifically that pressing the "save" button is necessary before clicking "done" at the top. We also plan to improve this workflow.

Thank you again for reporting the bug! We should have a fix live in Canvas in the next week.

ybader, 2022-04-22:
Hi Derek,

Thanks for the update, it's great to hear that you were able to reproduce the problem. 

The problem with the ingoing/outgoing labels being incorrect explains why I have been running into repeated splice failures, however is there a reason why this problem doesn't seem to occur to every model I have tried to print? The only reason I managed to find this problem myself was because one of the filaments I have requires relatively extreme splice settings (-1,2,-1), and I have found that while this problem exists and causes splice settings to incorrectly be entered into Canvas, the Palette occasionally does not run into problems and does occasionally complete prints using this troublesome filament without issue. Could it just be pure luck of what order of which filaments are being spliced in the project and some filaments combinations being more forgiving to incorrect splice settings?

brandonbloch, 2022-08-05:
Hi @ybader, apologies on the delay here. This bug has been resolved for some time now and should be working as expected once again. I'm going to close this issue, but if you're still having problems please feel free to comment here/re-open the issue.

Cheers,
Brandon

closed 2022-08-05 by brandonbloch