MosaicManufacturing/canvas-issues #237: 3MF support for Canvas3D.io

GavBak, 2022-05-03:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

Would like to use 3mf files, canvas throws an error when I upload.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Update canvas to utilize 3mf

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Piddey, 2022-06-23:
3MF file support would be great! STL is a little bit outdate, isn't it?

GavBak, 2022-06-23:
3MF is what many people slice to now, without 3MF compatibility there are many files we cannot use in canvas.

HauserGer, 2023-04-05:
Agreed! Can't even believe it does not support 3MF files. I am returning my Palette 3 Pro due to this, not joking.

Bloomingdale-Robotics, 2023-10-07:
Need 3MF support

Strell69, 2023-10-18:
There are Tribe files that use .3mf with colors already added saving lots of time having to color manually in Canvas. Adding .3mf is very useful.

moberli, 2023-11-25:
Agree - 3mf is coming up more and more..