MosaicManufacturing/canvas-issues #258: Accessory Mode

Fargo666, 2022-12-12:
**Describe the bug**
Hello,

Sorry but I am really and so ... disapointed by Pallette 3 Pro I just bought. 

So much disappointed, that I tried to use the accessory mode... and then ... as soon as the first step (device definition in Canvas hub showed in youtube video ), I gave up . Just because the option of declaring my devices as non connected ( without USB) seems to be "not existing anymore" (?!?!!!)

Am I wrong ? or is there an other tip, or whatever trick that could help me ?

Stephane
**To Reproduce**
Just Follow Accessory mode with Palette 3 as it is step by step described here : https://www.youtube.com/watch?v=X7NCA-7_ghg&t=7s
and there : https://support.mosaicmfg.com/Guide/Accessory+Mode+with+Palette+3/172?lang=en

**Expected behavior**
Just see Accessory mode with Palette 3 as it is step by step described here : https://www.youtube.com/watch?v=X7NCA-7_ghg&t=7s
and there : https://support.mosaicmfg.com/Guide/Accessory+Mode+with+Palette+3/172?lang=en

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Operating System (please complete the following information):**
 - OS: Windows 10
 - Chrome
https://canvas3d.io/

**Additional context**
Add any other context about the problem here.

brandonbloch, 2022-12-12:
Hi @Fargo666, thanks for reaching out! Apologies for the confusion caused — it looks like the video in this article hasn't been updated with some of the workflow improvements we've made in Canvas over the last few months. Our Customer Success team has been made aware of the issue and will be working on updating this article to avoid any further confusion.

Rest assured, however, that Accessory Mode has not been and will not be removed! In fact, despite the confusion, I hope you'll agree that the new approach is actually much simpler to set up and manage than what was displayed in that video. The article linked to in Step 1 ([mm3d.co/p3-canvas](http://mm3d.co/p3-canvas)) does have the correct up-to-date setup instructions (including how to enable Accessory Mode), so I would recommend following that guide directly.

brandonbloch, 2022-12-12:
(re-posting this reply to include some additional information)

Hi @Fargo666, thanks for reaching out! Apologies for the confusion caused — it looks like the video in this article hasn't been updated with some of the workflow improvements we've made in Canvas over the last few months. Our Customer Success team has been made aware of the issue and will be working on updating this article to avoid any further confusion.

Rest assured, however, that Accessory Mode has not been and will not be removed! In fact, despite the confusion, I hope you'll agree that the new approach is actually much simpler to set up and manage than what was displayed in that video. The article linked to in Step 1 ([mm3d.co/p3-canvas](http://mm3d.co/p3-canvas)) does have the correct up-to-date setup instructions (including how to enable Accessory Mode), so I would recommend following that guide directly.

Basically, projects are no longer tied directly to a particular device you have linked to your Canvas account (in this case, Palette 3 Pro), but instead you simply choose that Palette model directly within the project. After slicing, you'll be able to send the project directly to any compatible device you have linked to your Canvas account (also your P3P in this case). You can also configure your account to use a Palette 3 Pro in Accessory Mode for all new projects by default, by going to Account Settings (under the menu icon in the top-right) > New Projects and editing the default device.

Hopefully this helps clarify things! I'll leave this issue open for visibility until we can confirm that the documentation has been updated. In the meantime, feel free to reply to this thread or get in touch directly at support@mosaicmfg.com if you run into any other issues getting set up.

Cheers and happy printing,
Brandon

Fargo666, 2022-12-13:
Ok, I got it !
Thank you very much,Brandon .  it's clearer now.
And at least, that issue made me more curious about Canvas that I underestimated at first glance, just because I was really disappointed in not having anymore the possibility to use my usual Simplify3D integrated to Chroma, as it seems to be possible with previous models of palette. 
Then, I will try to progress with Canvas that finally is not so rough that I presumed ...

Have a nice day 
Stephane

closed 2023-02-14 by brandonbloch