MosaicManufacturing/canvas-issues #263: Can't open any projects, "Oops, Something went wrong (js error: Cannot read properties of undefined (reading 'toLowerCase'))

dgrant, 2023-02-05:
**Describe the bug**

When I click on any project, or create a new project, I get this "Oops, something went wrong" error when it tries to open.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to "Projects"
2. Double-click on any project
3. See error

**Expected behavior**
I expected to see my project open up

**Screenshots**

[image]

[image]

**Operating System (please complete the following information):**
 - OS: Windows
 - Browser: Chrome
 - Version: 109.0.5414.120 (Official Build) (64-bit)

**Additional context**

I haven't logged in in a long time.

brandonbloch, 2023-02-13:
Hi @dgrant, sounds like a possible issue with a rather old printer or style profile. Would you be willing to get in touch with our support team at support@mosaicmfg.com with some of the following info (rather than needing to post it here):

- your account's username or email address
- the URL of one of the projects with this issue

Thanks,
Brandon

xrayjoe, 2023-02-21:
I'm having the same issue. What is the solution?

This is very upsetting as I took a break from using the Palette and now it's useless without the website working. Whatever happened to the standalone software??

brandonbloch, 2023-02-21:
Hi @xrayjoe, we've been making improvements to better avoid these issues but as mentioned above would you mind getting in touch at support@mosaicmfg.com with the info above? We're not sure yet what exactly causes this issue for some but not all users, so taking a look at an affected project should help us narrow down the root cause of this particular issue and verify if our improvements will effectively help to avoid these types of issues in the future.

As for standalone software, [Chroma](http://mm3d.co/chroma) is still available with support for all Palette and Palette 2 models, and [p2pp](https://github.com/tomvandeneede/p2pp) is available as a community-developed PrusaSlicer plugin with support for all Palette 2 and Palette 3 models.

Thanks,
Brandon